MallocArray/airgradient_esphome #109: Error "ID EC:DA:3B:XX:XX:XX-sensor-pm_2_5 is already used by sensor.ag_open_air_o_1pst_pm_2_5"

Qhilm, 2025-01-12:
>[!NOTE]
>ESPhome: 2024.12.2
>Home Assistant: 2025.1.2
>Hardware: Open Air O 1PST

I am not sure when this started and I'm also not sure this is related to `airgradient_esphome` and not an issue with ESPhome itself.

Error:

>Logger: homeassistant.components.sensor
>Source: helpers/entity_platform.py:773
>integration: Sensor (documentation, issues)
>First occurred: 15:50:29 (1 occurrences)
>Last logged: 15:50:29
>
>Platform esphome does not generate unique IDs. ID EC:DA:3B:XX:XX:XX-sensor-pm_2_5 is already used by >sensor.ag_open_air_o_1pst_pm_2_5 - ignoring sensor.ag-open-air-o-1pst_pm_2_5

It seems like ESPhome it is indeed probably the same sensor in both cases and HA somehow identifies it as different names. It appears that some hyphens are replaced by underscores, maybe it's a parsing issue.

Does that ring a bell for anyone?

```yaml
substitutions:
  name: "ag-open-air-o-1pst"
  friendly_name: "AG Open Air O-1PST"
  name_add_mac_suffix: "false"  # Must have quotes around value

# Enable Home Assistant API
api:  # Add encryption key as desired
  encryption:
    key: !secret home_assistant_encryption_key

# web GUI not needed unless troubleshooting
# web_server:

ota:  # Add password as desired
  password: !secret ota_password
  platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  reboot_timeout: 15min
  ap:
    ssid: "${friendly_name} Hotspot"
    password: !secret fallback_ssid_password

dashboard_import:
  package_import_url: github://MallocArray/airgradient_esphome/airgradient-open-air-o-1pst.yaml
  import_full_config: false

packages:
  board: github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board.yaml
  # captive_portal: github://MallocArray/airgradient_esphome/packages/captive_portal.yaml
  # pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003t.yaml
  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003t_extended_life.yaml
  co2: github://MallocArray/airgradient_esphome/packages/sensor_s8.yaml
  tvoc: github://MallocArray/airgradient_esphome/packages/sensor_sgp41.yaml
  airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_esp32-c3.yaml
  hardware_watchdog: github://MallocArray/airgradient_esphome/packages/watchdog.yaml
  wifi: github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime: github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml
```

MallocArray, 2025-01-13:
That is an odd error and not one I've seen.  Just a few days ago I pushed 4.1.0 which does change the framework to esp-idf.  Do you know if this started with that change, or was this prior to January 10th?

Depending on how important sensor history is to you, you could go in HA and find any sensor entries with `_pm_2_5` in the name and try removing them so the new entry can be imported.

Not sure why it is trying to use a device name that looks like a MAC address instead of your `name: ` entry, but I don't see anything off in your config, like bluetooth_tracker that I haven't tested.

If you want to try a few things, you could attempt changing the package `pm_2_5` to the non-extended life version just for troubleshooting.
You could also try changing the `board:` line to end with `airgradient_esp32-c3_board-arduino.yaml` to go back to the arduino framework if this problem only started after compiling beginning Jan 10th

Qhilm, 2025-01-28:
ah, it appears you switched from aduino to esp-idf in 4.1.0 and that can cause issues such as what I described in the previous post:

https://github.com/Blackymas/NSPanel_HA_Blueprint/issues/1821

I will reflash it over usb and see how it goes.

Qhilm, 2025-01-28:
I personally prefer esp-idf, but I agree it's a significan challenge.

At the same time, I am surprised I'm the first one reporting this. Maybe it's a rare occurrence overall and we're both unfortunate.

MallocArray, 2025-01-28:
I saw that, an individual users can apply that to their main file, but it isn't something I want to include in the files here.  Trying to avoid any extra files that have to be downloaded, and I read through the thread but wasn't sure if it was a one-time fix, as I understood it to be needed until you did the serial flash.

Qhilm, 2025-01-29:
Maybe you're right, the thread isn't clear on this actually, I misread it. Even if it was a one off, I'm not sure it would be usable as a mass solution or all users of this repo.

I now think going back to arduino and wait for bootloader to be flashable OTA (https://github.com/esphome/esphome/pull/5535) is the better approach. Then switching frameworks can be done OTA properly, in one release.

Just my view though. I'm happy to test a couple of things though.

MallocArray, 2025-02-21:
I updated the README to mention this Breaking Change of needing to do one flash over USB to finish the migration

Thanks for bringing this to my attention

closed 2025-02-21 by MallocArray