MallocArray/airgradient_esphome #111: Open Air Temperature Differential

spiceywasabi, 2025-01-28:
Hello,

I just flashed my AirGradient Open Air 1-PPT. This version has the two temperature sensors averaging out. I noticed the temperature average was 41 degrees F instead of 60 F. Quick look at the data shows one of the sensors is reporting F but is actually in C. Looking quickly at the full configuration it seems everything is normalized to C so I'm unsure if this is a bug here or somewhere else. 

[image]

Configuration used:
```yaml
dashboard_import:
  package_import_url: github://MallocArray/airgradient_esphome/airgradient-open-air-o-1ppt.yaml
  import_full_config: 

packages:
  board: github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board.yaml
  captive_portal: github://MallocArray/airgradient_esphome/packages/captive_portal.yaml
  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003t.yaml
  # pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003t_extended_life.yaml
  pm_2.5_2: github://MallocArray/airgradient_esphome/packages/sensor_pms5003t_2.yaml
  # pm_2.5_2: github://MallocArray/airgradient_esphome/packages/sensor_pms5003t_2extended_life.yaml
  tvoc: github://MallocArray/airgradient_esphome/packages/sensor_sgp41.yaml
  airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml
  hardware_watchdog: github://MallocArray/airgradient_esphome/packages/watchdog.yaml
  wifi: github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime: github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml

```

MallocArray, 2025-01-28:
That is quite odd.  I looked over the average and I don't see anything that would cause it to treat one of them as C and one as F

```yaml
    lambda: return (id(temp).state + id(temp_2).state) / 2.0;
    unit_of_measurement: °C
```

I don't have this particular model so I can't see if mine is behaving the exact same.  Any chance the fans in one of the sensors has failed or been stopped up by bugs so might not be circulating?  It is a pretty wide swing between them and I would expect it to be that far apart.

You could check in ESPHome using the Logs feature to see what is being reported by the sensors directly, before any corrections or averages.  It should look something like:
`[15:54:51][D][pmsx003:285]: Got PM1.0 Concentration: 2 µg/m^3, PM2.5 Concentration 5 µg/m^3, PM10.0 Concentration: 5 µg/m^3, Temperature: 15.4°C, Humidity: 23.7%`

If one is just straight up reporting widely different values for the raw temp in C, then we narrowed it down to the sensor itself reporting something off.

Note: There is also a new report that some devices are not updating their partitions since the latest release changed to ESP-IDF over the Arduino framework.  I'm still looking into it and wouldn't expect it to be factoring into your issue, but if you want to try rolling back, you could change the board line to end with `airgradient_esp32-c3_board-arduino.yaml` to use the previous framework.

spiceywasabi, 2025-01-29:
I had it outside and it actually got warmer the colder it got outside. So its a faulty sensor. Sorry for the ticket! I'll go ahead and close it now. 

[image]

closed 2025-01-29 by spiceywasabi