MallocArray/airgradient_esphome #112: Compiler deprecation warning for LED Strip

cryptonym64, 2025-02-15:
Using the package files from repo...

Compiling .pioenvs/airgradient_one/src/main.cpp.o
In file included from src/esphome/components/esp32_rmt_led_strip/led_strip.h:12,
                 from src/esphome.h:31,
                 from src/main.cpp:3:
/data/cache/platformio/packages/framework-espidf/components/driver/deprecated/driver/rmt.h:18:2: warning: #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h" [-Wcpp]
   18 | #warning "The legacy RMT driver is deprecated, please use driver/rmt_tx.h and/or driver/rmt_rx.h"
      |  ^~~~~~~

MallocArray, 2025-02-15:
I had looked this one up and it was a known thing in ESPHome due to the version of ESP-IDF that is currently being used.  The issue is addressed in newer ESP-IDF versions, but ESPHome is a few versions behind due to compatibility.

The recent changes I made switched from Arduino framework to ESP-IDF which is causing these to be visible, but currently it is only a deprecation warning

cryptonym64, 2025-02-16:
> I had looked this one up and it was a known thing in ESPHome due to the version of ESP-IDF that is currently being used. The issue is addressed in newer ESP-IDF versions, but ESPHome is a few [...]

I appreciate it being moved to ESP-IDF and making room for Bluetooth proxy to be added.

MallocArray, 2025-02-16:
https://github.com/esphome/issues/issues/5357

closed 2025-02-16 by MallocArray