MallocArray/airgradient_esphome #113: arithmetic deprication warning on sht4x.cpp.o

cryptonym64, 2025-02-15:
using files from repo...

Compiling .pioenvs/airgradient-one/src/esphome/components/sht4x/sht4x.cpp.o
src/esphome/components/sht4x/sht4x.cpp: In member function 'virtual void esphome::sht4x::SHT4XComponent::setup()':
src/esphome/components/sht4x/sht4x.cpp:22:63: warning: arithmetic between enumeration type 'esphome::sht4x::SHT4XHEATERTIME' and floating-point type 'float' is deprecated [-Wdeprecated-enum-float-conversion]
   22 |     uint32_t heater_interval = (uint32_t) (this->heater_time_ / this->duty_cycle_);
      |                                            ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~

MallocArray, 2025-02-15:
Thanks for reporting this.  I'm pretty sure this is coming from the ESPHome module directly and nothing that is being done specifically in this repo.  I'll see if there is an open Issue there to reference here

MallocArray, 2025-02-16:
Being tracked in ESPHome under
https://github.com/esphome/issues/issues/6708

closed 2025-02-16 by MallocArray