MallocArray/airgradient_esphome #114: esp32_rmt_led_strip Error

Furrypaw, 2025-02-19:
INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/ag-one.yaml...
Failed config

light.esp32_rmt_led_strip: [source /data/packages/76b4402f/packages/led.yaml:3]
  platform: esp32_rmt_led_strip
  rgb_order: GRB
  pin: GPIO10
  num_leds: 11
  
  This feature is not available for the IDF framework version 5.
  rmt_channel: 0
  chipset: ws2812
  name: LED Strip
  id: led_strip
  restore_mode: RESTORE_DEFAULT_OFF

cryptonym64, 2025-02-19:
Updated my ESPHome Device Builder to 2025.2.0 and receive same.

MallocArray, 2025-02-19:
Looks like the new update negates the need for specifying a channel so I'll test removing that.

https://github.com/esphome/esphome/pull/7770

cryptonym64, 2025-02-19:
I believe the fix is to remove the "rmt channel" line in the attribute list as it has been obsoleted in v5.

refer to https://github.com/esphome/esphome-docs/pull/4480

components/remote_transmitter.rst	Marked rmt_channel as obsolete with IDF framework v5;

For time being, I simply copied content of the led.yaml into my base configuration file, placed \# before the rmt_channel line and commented out the line referring to the led.yaml in the repo.

MallocArray, 2025-02-21:
Just uploaded 5.0.0 which includes a fix for this using ESPHome 2025.2.0

Do note that if you are coming from something earlier than 4.1.0 of this config, please do one flash over USB to do a full format of the internal storage so ESP-IDF is fully migrated.

closed 2025-02-21 by MallocArray

cryptonym64, 2025-02-22:
@MallocArray thanks for the quick fix. Working fine.