MallocArray/airgradient_esphome #118: Failure updating with latest YAML

ih8gates, 2025-02-21:
I initially posted the info below to the [AirGradient forums](https://forum.airgradient.com/t/failure-updating-with-latest-mallocarray-yaml-in-esphome/3708). When I returned here to create an issue, I discovered that there had been some updates. I tried my update again with the newest full_config YAML for my AGOne and got a different error.

```
Failed config

font: [source /config/esphome/air-gradient-living-room.yaml:520]
  - 
    'family' is a required option for [file].
    file: 
      
      [path] is an invalid option for [file]. Please check the indentation.
      path: /config/.esphome/font/Open Sans@400@False@v1.ttf
      type: gfonts
    id: open_sans_14
    size: 14
    glyphs: 
      - !"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³
    glyphsets: []
    ignore_missing_glyphs: False
    bpp: 1
```

**Initial issue:**

I got a notice (in Home Assistant) that there was an update for my AGOne. When I tried to update it, it failed, so I visited MallocArray's Git repository to attempt to fix things. 

I copied [the correct YAML from the repository's full_config folder](https://github.com/MallocArray/airgradient_esphome/blob/main/full_config/ag-one.yaml) and updated my personal customizations (name, friendly name, wifi, encryption key), leaving everything else as-is. When I try to install, I get the following error:

`UnknownPackageError: Could not find the package with '~3.40407.0 @ *' requirements for your system 'linux_aarch64'`

Is this an issue with YAML or something I need to fix on my end?

After this failed, I went to the ["airgradient-one.yaml" in the main folder of the repository](https://github.com/MallocArray/airgradient_esphome/blob/main/airgradient-one.yaml), again adding my personal bits to the YAML, but I get an error with that (I'm pretty sure this is the same error I got when initially trying to update):

```
Failed config

light.esp32_rmt_led_strip: [source /data/packages/76b4402f/packages/led.yaml:3]
  platform: esp32_rmt_led_strip
  rgb_order: GRB
  pin: GPIO10
  num_leds: 11
  
  This feature is not available for the IDF framework version 5.
  rmt_channel: 0
  chipset: ws2812
  name: LED Strip
  id: led_strip
  restore_mode: RESTORE_DEFAULT_OFF
```

MallocArray, 2025-02-21:
There are a couple of things going on here.

The first issue you mentioned with the font and stuff about file and family looks like they changed something in ESPHome related to how the full configs are generated, as in earlier versions it looked like
```yaml
font:
- file:
    family: Open Sans
    weight: 400
    italic: false
    refresh: 1d
    type: gfonts
  id: open_sans_14
  size: 14
```
but with this newest version, it is significantly different:
```yaml
font:
- file:
    path: /config/.esphome/font/Open Sans@400@False@v1.ttf
    type: gfonts
  id: open_sans_14
  size: 14
  glyphs:
  - '!"%()+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/µ³'
  glyphsets: []
  ignore_missing_glyphs: false
  bpp: 1
  extras: []
```

I can try to make a minimal example to open an Issue with the ESPHome project about it.  Unfortunately I found out that I had been using an older ESPHome version when doing the config exports, so it isn't just comparing the last release to this one, but could be something that changed in the last several months.

This should only be impacting the Full Config files, and I would highly recommend using the files at the root of the project which reference individual packages, rather than the full configs that combine them all into a single file.

---

The issue I believe you are running into with the `airgradient-one.yaml` at the root of the project is due to how long ESPHome caches packages by default. It will not pull down updated packages until what you have is 1 day old, so since you were working on it earlier today, I expect that it is not pulling down the most recent changes I pushed up.  If you wait 24 hours, then I believe it will work as expected.

You could temporarily change the line for the LED to a more verbose version and change the `refresh:` value to something lower, but I didn't want to standardize on that for the regular files.  If you are using the straight Docker container for ESPHome and not the HA addon, there are other ways to delete the files from your local repository.
https://esphome.io/components/packages.html#configuration-variables

I think it would look like this instead of the current line starting with `led:`
```yaml
  led:
    url: https://github.com/MallocArray/airgradient_esphome/tree/main/packages
    files: [led.yml]
    ref: main
    refresh: 1h
```

Or it might need to be 
```yaml
  led:
    url: https://github.com/MallocArray/airgradient_esphome
    files: [packages/led.yml]
    ref: main
    refresh: 1h
```

ih8gates, 2025-02-21:
I was able to update with the minimal config version (which wasn't working for me this morning).

MallocArray, 2025-02-21:
https://github.com/esphome/issues/issues/6799

Opened an issue about the font issue with full config

MallocArray, 2025-02-27:
I manually replaced the auto-generate font section that is not functional with working configs, so the full_config file for this version should be working.

Looks like the esphome team is aware and looking into it.