MallocArray/airgradient_esphome #119: Can not build after upgrade to ESPHome 2025.2.1

rkuijer, 2025-02-26:
The build fails after the upgrade to ESPHome 2025.2.1 during the initial build steps.
It's unclear to me if it's related to the airgradient_esphome or a library it's depending on. 

`
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:505 (__project):
`
full logging include

[logs_test_compile.txt](https://github.com/user-attachments/files/18981685/logs_test_compile.txt)

MallocArray, 2025-02-26:
That looks like an issue with your ESPHome instance, rather than with this configuration file.  I upgraded my personal instance to 2025.2.1 and was able to install successfully.

You could try using the "Clean Build Files" option and/or restarting ESPHome, which depends on how you have it installed, such as an HA addon or standalone container.

[image]

rkuijer, 2025-02-27:
Very strange, I already tried the clean function without success. This was the only ESPHome device that failed to build.
Today I removed the ESPHome addon from HA and reinstalled it. 
After that everything compiled with success. 
So this issue can be closed. 

Thanks for the quick response and the suggestions.