MallocArray/airgradient_esphome #125: rmt_channel is a required option for the Arduino framework.

kroeckx, 2025-04-01:
Since ESPHome 2025.3 update doesn't work anymore. I get the following error:
```
INFO ESPHome 2025.3.3
INFO Reading configuration /config/esphome/esphome-web-12e5a0.yaml...
Failed config

light.esp32_rmt_led_strip: [source /data/packages/76b4402f/packages/led.yaml:3]
  
  rmt_channel is a required option for the Arduino framework.
  platform: esp32_rmt_led_strip
  rgb_order: GRB
  pin: 10
  num_leds: 11
  chipset: WS2812
  name: LED Strip
  id: led_strip
  restore_mode: RESTORE_DEFAULT_OFF
  disabled_by_default: False
  gamma_correct: 2.8
  default_transition_length: 1s
  flash_transition_length: 0s
```

It seems that in commit b2a2814a6b86b4681f16395f752ffd4f8b583d20, rmt_channel was removed, claiming that was needed for 2025.2.0.

kroeckx, 2025-04-01:
My config looks a lot like airgradient-one.yaml. Did that used to contain type: arduino?

MallocArray, 2025-04-01:
This file: `https://github.com/MallocArray/airgradient_esphome/blob/main/airgradient-one.yaml` references `github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board.yaml` which is now using esp idf.  So when you install, it should reach out to pull the latest versions and use it.

Maybe post your full YAML, removing any encryption or passwords

MallocArray, 2025-04-01:
Perfect thanks.

Under substitutions, you don't need `config_version: 2.0.1` anymore, as it comes in with the board so the version updates appropriately.  It doesn't break anything, but will confuse a human.

The root cause is this in your YAML
```
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino
```
This is overriding what is set in airgradient_esp32-c3_board.yaml which changed to IDF, but you have it set to arduino, which makes it incompatible with the `led.yaml`

So easiest solution is to remove that section from your yaml, which will switch you to IDF.  Do note that it is highly recommended to do one firmware install from a USB cable so it can format the internal storage to be compatible with IDF.  If you don't, it may randomly boot back into the old config.  Once it is done, you can do over the air going forward as long as you stay with IDF.

If you want to stick with Arduino, then I would still remove that section, but change these two lines under packages to be this:
```yaml
  board: github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board-arduino.yaml
  led: github://MallocArray/airgradient_esphome/packages/led.yaml@4.0.7
```

kroeckx, 2025-04-01:
They have all been updated.

closed 2025-04-01 by kroeckx