MallocArray/airgradient_esphome #126: LED lights return to green every few seconds

madkatz01, 2025-04-06:
The LED lights on my AirGradient One (Model I-9PSL) just return to green after a few seconds. I can manually turn them on and off fine, and change the color in Home Assistant - but after a few seconds, they default back to green (although HA seems to think the color is still what I set it at, i.e. blue).

Anyone else having this issue?

Running Firmware 5.1.0

madkatz01, 2025-04-06:
Ah, was able to remedy this by just commenting out the ledCO2 yaml. Now I can manually control the colors with automations in HAOS.

closed 2025-04-06 by madkatz01

MallocArray, 2025-04-06:
What ESPHome version are you running? I've been keeping up with them except for the most recent release and I haven't seen this behavior, although I use the Combo LED package instead of just CO2.
I'll try switching mine to CO2 and see if I experience the same.

Any chance that the CO2 values on the display are changing significantly, such as going to NaN or 0?

madkatz01, 2025-04-06:
Running everything in HAOS, including ESPhome and should all be updated. I am still having trouble - with the LEDCO2 yaml commented out, I can control the LED colors (and on/off), but have lost the ability to control brightness and fade - even though I see those features specified in the regular LED yaml. 

Sensors on the device seem appropriate and stable (compared to my Apollo Air-1).

MallocArray, 2025-04-06:
Perhaps I should clarify what the default CO2 package for LED does.  When the CO2 levels are near 400, it will be true green. As the value moves to 1000 it will slowly transition to Yellow, but if you are between 400 and 1000, it will be proportionally between the two.
Once it is over 1000, it will start transitioning to Red as it nears 2000.  So if your values are around 600-800, it will still be a shade of green by default.  It checks every 5 seconds and adjusts the colors, which is why you can change it via HA, but it goes back to the color based on the values.

By removing the led_co2 package, there isn't anything that is changing the brightness or fade, it is just doing the default behavior of setting the color.  There are sliders and boxes for the brightness and fade, but without the rest of the package to tell it how to use those, it doesn't do anything.

You might be able to manually define the light section and add the brightness option, but it would take a bit more work for the fade.

If you want to try this out, I'll suggest commenting out the led.yaml as well, and then at the bottom of your config file, add this, which may work in theory, but I haven't tried it out.

```yaml
light:
    # https://esphome.io/components/light/esp32_rmt_led_strip.html
  - platform: esp32_rmt_led_strip
    rgb_order: GRB
    pin: GPIO10  # Pin 16
    num_leds: 11
    chipset: ws2812
    name: "LED Strip"
    id: led_strip
    restore_mode: RESTORE_DEFAULT_ON
    color_brightness: !lambda 'return ((id(led_brightness).state) / 100.0);'
```