MallocArray/airgradient_esphome #135: Poor uptime (~450s) for just one AG unit of several

roens, 2025-04-20:
This may be related to #36…

I may be experiencing this issue…

I've got several AirGradient units here: an outdoor model with two PMS5003 sensors, two simple indoor ones, and a "pro" (v9). Only the pro model has been experiencing this super short uptime reboot loop, very reliably ~450s between reboots:

[image]
<br><br>

The v9 board I've got in my AG Pro is based on an `ESP32-C3-Mini-1`. I believe the short uptime started after I migrated it to being managed by the Home Assistant ESPHome Add-on, on April 2. Previously I was compiling & uploading its firmware from my (macOS) laptop. At the time of the switch its uptime was about 11 days.

[image]
<br><br>

I also use Ruckus, with two APs. The network all of my IoT equipment is on is just 2.4Ghz. This AG Pro is nearer to one of the two APs, and is reported as having a good signal.

I've tried "turning off" both the AirGradient API reporting and the Captive Portal, but that hasn't made a difference. Also the "config button" GPIO pin wasn't lining up (preventing compilation) and haven't bothered to figure out what that should be, so the chunk at the end is commented out.

And in spite of this short uptime, HA is getting some data from the unit: temp, RH, CO2, PM, and VOC.

```yaml
# AirGradient Pro V3.3 - V4.2
# https://www.airgradient.com/open-airgradient/instructions/overview/

substitutions:
  name: "diningroom-ag-pro"
  friendly_name: "Dining Room AG Pro"
  name_add_mac_suffix: "false"  # Must have quotes around value

esphome:
  comment: 'https://github.com/MallocArray/airgradient_esphome'

# Enable Home Assistant API
api:
  encryption:
    key: !secret home_assistant_encryption_key

ota:
  platform: esphome
  password: !secret ota_password

wifi:
  use_address: "${name}.hoem"
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
  reboot_timeout: 15min
  domain: '.hoem'

dashboard_import:
  package_import_url: github://MallocArray/airgradient_esphome/airgradient-pro.yaml
  import_full_config: false

packages:
  board:          github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board.yaml
# captive_portal: github://MallocArray/airgradient_esphome/packages/captive_portal.yaml
  pm_2.5:         github://MallocArray/airgradient_esphome/packages/sensor_pms5003_extended_life.yaml
  co2:            github://MallocArray/airgradient_esphome/packages/sensor_s8.yaml
  temp_humidity:  github://MallocArray/airgradient_esphome/packages/sensor_sht40.yaml
  tvoc:           github://MallocArray/airgradient_esphome/packages/sensor_sgp41.yaml
  aqi:            github://MallocArray/airgradient_esphome/packages/sensor_nowcast_aqi.yaml
  display:        github://MallocArray/airgradient_esphome/packages/display_sh1106_single_page.yaml
# airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_esp32-c3.yaml
  config_button:  github://MallocArray/airgradient_esphome/packages/config_button.yaml
  wifi:           github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime:         github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml

#binary_sensor:
#  - id: !extend config_button
#    pin:
#      number: D7
```

roens, 2025-04-27:
ok. 🤦🏻

I added in the package:

    watchdog:       github://MallocArray/airgradient_esphome/packages/watchdog.yaml

And it no longer is restarting every ~450s. Thanks for that added package. 😆

closed 2025-04-27 by roens

MallocArray, 2025-04-27:
I see the issue. The v9 board isn't considered a Pro model, it is called the ONE. 
Your file is based on the older Pro with the D1 chip, but you have a few lines updated. 

Compare your config with the airgradient_one.yaml file to get all of the standard stuff. 
You'll also be able to enable the LED bar once you add that package and turn it on in HomeAssistsnt

roens, 2025-05-09:
Indeed! I had an older AG Pro, that I upgraded by swapping the main board inside with their newer v9 One board.

(It's been continuing to work well since adding in the watchdog module)