MallocArray/airgradient_esphome #139: No long term stats in HA with AQI?

pyrodex, 2025-05-17:
I noticed the other entities seem to have long term stats in my HA but I don't see this for AQI. I only have the last 3 days, my recorder settings, so not sure why this entity isn't doing long term stats.

Record settings for example:

```
recorder:
  purge_keep_days: 3
  exclude:
    domains:
      - uptime
      - time_date
      - worldclock
    entity_globs:
      - sensor.time*
      - sensor.uptime*
      - sensor.*_distance_to_*
```

MallocArray, 2025-05-18:
I'm not seeing anything that looks like it would trigger your 3 day purge.  It is a Template Sensor that is calculating AQI with `device_class: aqi` assuming you are using the package `sensor_pms5003.yaml`

Might need to open an Issue with HomeAssistant or reach out on their Discord to see why it appears to be getting purged early.

I haven't done anything specific with long term stats, so this whole `recorder` thing is new to me, but I would be interested in how to get HomeAssistant to keep certain metrics longer.

MallocArray, 2025-05-26:
Did you open a case with ESPHome to have them give any feedback?

pyrodex, 2025-05-26:
I have not, sorry real life.