MallocArray/airgradient_esphome #140: Update to support breaking changes in ESPHome 2025.5.0 with HTTP requests.

pyrodex, 2025-05-21:
According to the release notes they made a breaking change around http request headers (https://esphome.io/changelog/2025.5.0.html#http-request)

Looks like all occurrences of `headers` will need to be replaced with `request_headers`. 

```
esphome: None
  name: ag-open-air-o-1ppt
  friendly_name: AG Open Air O-1PPT
  name_add_mac_suffix: False
  project: 
    name: mallocarray.airgradient
    version: 5.1.0
  min_version: 2025.2.0
  on_boot: 
    - priority: 200
      then: 
        if: 
          condition: 
            switch.is_on: upload_airgradient
          then: 
            - http_request.post: 
                url: !lambda |-
                  return "https://hw.airgradient.com/sensors/airgradient:" + get_mac_address() + "/measures";
                
                The 'headers' options has been renamed to 'request_headers'.
                headers: 
                  Content-Type: application/json
                json: 
                  wifi: !lambda |-
                    return to_string(-50);
  build_path: build/ag-open-air-o-1ppt
```

MallocArray, 2025-05-21:
Addressing with #141

MallocArray, 2025-05-21:
This should be addressed now, but I am seeing a new warning with 2025.5.0 that doesn't seem to be breaking anything, but be aware:

`[12:26:12][W][pmsx003:125]: Start character 1 mismatch: 0x00 != 0x42`

https://github.com/esphome/issues/issues/7040