MallocArray/airgradient_esphome #146: Can't install after last ESPHome update to 2025.5.0

Motya-HA, 2025-05-22:
[logs_esphome-web-0c0acc_run.txt](https://github.com/user-attachments/files/20382295/logs_esphome-web-0c0acc_run.txt)

MallocArray, 2025-05-22:
This was addressed with 5.2.0 yesterday.  Likely your ESPHome has a cached version of the packages that it keeps by default for 24 hours.  If you try again in a few hours it may start picking up the latest changes. 

If you are running ESPHome from a standalone Docker container, you can go into your config folder and remove the .esphome/packages folder and then it will force it to pull down the latest.

If using the Add-on in HomeAssistant, it is just easiest to wait.

Motya-HA, 2025-05-22:
Actually in HA Add-on it was working, but not in standalone Docker
container

I'll try and remove the .esphome/packages folder

MallocArray, 2025-05-26:
Is it working now?