MallocArray/airgradient_esphome #147: Deprecate PMS5003 extended life packages

MallocArray, 2025-05-26:
The only difference in the extended life packages is a substitution that lets a user overwrite the update interval named `pm_update_interval`

Add this same substitution to the main packages with the same default as the existing packages, and then users can set their own update interval with a single package.  Several less packages to maintain that have almost identical information.

Will have to figure out how to communicate this to users.  Considering adding to release notes that it will be deprecated in the future and then in a few months removing the package from the repo.  Then when users try to install after that date, it will throw an error that will bring them back to the repo to see the change if they haven't yet.

MallocArray, 2025-08-22:
Waiting on new fixes from the ESPHome project to fully resolve the current issues with the pmsx003 component

https://github.com/esphome/issues/issues/7063
https://github.com/esphome/issues/issues/7040

cryptonym64, 2026-11-01:
Did the blockers get addressed in https://github.com/esphome/esphome/pull/14832 ?