MallocArray/airgradient_esphome #150: airgradient_api_esp32-c3_dual_pms5003t.yaml: error: 'String' was not declared in this scope; did you mean 'ArduinoJson6185_71::String'?

ex-nerd, 2025-06-13:
I'm getting a compile error with the latest ESPHome (whatever version the HA-OS add-on uses).

```
/data/packages/76b4402f/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml: In lambda function:
/data/packages/76b4402f/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml:22:7: error: 'String' was not declared in this scope; did you mean 'ArduinoJson6185_71::String'?
   22 |                 String jsonString;
      |       ^     
      |       ArduinoJson6185_71::String
In file included from .piolibdeps/airgradient-open-air/ArduinoJson/src/ArduinoJson/Strings/Adapters/JsonStringAdapter.hpp:8,
                 from .piolibdeps/airgradient-open-air/ArduinoJson/src/ArduinoJson/Strings/StringAdapters.hpp:8,
                 from .piolibdeps/airgradient-open-air/ArduinoJson/src/ArduinoJson/Memory/MemoryPool.hpp:10,
                 from .piolibdeps/airgradient-open-air/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:7,
                 from .piolibdeps/airgradient-open-air/ArduinoJson/src/ArduinoJson/Variant/SlotFunctions.hpp:8,
                 from .piolibdeps/airgradient-open-air/ArduinoJson/src/ArduinoJson/Array/ArrayIterator.hpp:7,
                 from .piolibdeps/airgradient-open-air/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from .piolibdeps/airgradient-open-air/ArduinoJson/src/ArduinoJson.hpp:17,
                 from .piolibdeps/airgradient-open-air/ArduinoJson/src/ArduinoJson.h:9,
                 from src/esphome/components/json/json_util.h:11,
                 from src/esphome/core/string_ref.h:11,
                 from src/esphome/core/entity_base.h:5,
                 from src/esphome/components/sensor/sensor.h:5,
                 from src/esphome/core/controller.h:17,
                 from src/esphome/components/api/api_server.h:11,
                 from src/esphome/components/api/api_connection.h:8,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
.piolibdeps/airgradient-open-air/ArduinoJson/src/ArduinoJson/Strings/String.hpp:9:7: note: 'ArduinoJson6185_71::String' declared here
    9 | class String {
      |       ^~~~~~
/data/packages/76b4402f/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml:58:26: error: 'jsonString' was not declared in this scope
   58 |                 serializeJson(doc, jsonString);
```

The file in question: https://github.com/MallocArray/airgradient_esphome/blob/main/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml#L22 (also copied to the [full config](https://github.com/MallocArray/airgradient_esphome/blob/main/full_config/ag-open-air-o-1ppt.yaml#L616))

Oddly, this all used to work fine. However, it looks like the JSON process as been significantly cleaned up in the single-sensor version of the file, so maybe there is some cleanup needed?

ex-nerd, 2025-06-15:
I can confirm it's `ESPHome 2025.5.2`

I did a bit more digging and switched to just using your stock [airgradient-open-air-o-1ppt.yaml](https://github.com/MallocArray/airgradient_esphome/blob/main/airgradient-open-air-o-1ppt.yaml)

I made two changes below:
- comment out `captive_portal`
- add `ap:` under `wifi:`

```yaml
# AirGradient Open Air Outdoor Monitor with dual PMS5003T sensors
# Model: O-1PPT
# https://www.airgradient.com/open-airgradient/instructions/overview/

substitutions:
  name: "ag-open-air-o-1ppt"
  friendly_name: "AG Open Air O-1PPT"
  name_add_mac_suffix: "false"  # Must have quotes around value

# Enable Home Assistant API
api:  # Add encryption key as desired

ota:  # Add password as desired
  platform: esphome
  # password:

wifi:
  ap:

dashboard_import:
  package_import_url: github://MallocArray/airgradient_esphome/airgradient-open-air-o-1ppt.yaml
  import_full_config: false

packages:
  board: github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board.yaml
  # captive_portal: github://MallocArray/airgradient_esphome/packages/captive_portal.yaml
  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003t.yaml
  # pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003t_extended_life.yaml
  pm_2.5_2: github://MallocArray/airgradient_esphome/packages/sensor_pms5003t_2.yaml
  # pm_2.5_2: github://MallocArray/airgradient_esphome/packages/sensor_pms5003t_2extended_life.yaml
  tvoc: github://MallocArray/airgradient_esphome/packages/sensor_sgp41.yaml
  airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml
  hardware_watchdog: github://MallocArray/airgradient_esphome/packages/watchdog.yaml
  wifi: github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime: github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml
```

I wasn't using `captive_portal.yaml` because I have my own wifi config inline. Apparently something in ESPHome throws this odd `'String' was not declared in this scope` error if you don't include the `captive_portal:` line somewhere in the config.

I guess it's safe to close this…

MallocArray, 2025-07-03:
This looks to be specific to the format of the http_request.post used with the dual PMS5003t API call only.

I just opened https://github.com/esphome/issues/issues/7194 to see what can be done to address it

MallocArray, 2025-10-18:
Addressed in https://github.com/MallocArray/airgradient_esphome/releases/tag/5.3.3

Depending on when you last tried to Install, you may have to wait 24 hours to get the updated files

closed 2025-10-18 by MallocArray