MallocArray/airgradient_esphome #160: PMS5003t sensor values are all "NA" on Open Air O 1PST

Qhilm, 2025-07-04:
Only happens when using `sensor_pms5003t.yaml`. If switching to `sensor_pms5003t_extended_life.yaml`, everything works.

I can't see any error in the logs though:

<details>
  <summary>logs</summary>

```
NFO ESPHome 2025.6.3
INFO Reading configuration /config/esphome/ag-open-air-o-1pst.yaml...
INFO Starting log output from 192.168.1.138 using esphome API
INFO Successfully resolved ag-open-air-o-1pst @ 192.168.1.138 in 0.000s
INFO Successfully connected to ag-open-air-o-1pst @ 192.168.1.138 in 0.014s
INFO Successful handshake with ag-open-air-o-1pst @ 192.168.1.138 in 0.077s
[10:54:08][I][app:137]: ESPHome version 2025.6.3 compiled on Jul  3 2025, 19:38:03
[10:54:08][I][app:139]: Project mallocarray.airgradient version 5.3.0
[10:54:08][C][wifi:613]: WiFi:
[10:54:08][C][wifi:434]:   Local MAC: [REDACTED]
[10:54:08][C][wifi:439]:   SSID: [redacted]
[10:54:08][C][wifi:442]:   IP Address: 192.168.1.138
[10:54:08][C][wifi:446]:   BSSID: [redacted]
[10:54:08][C][wifi:446]:   Hostname: 'ag-open-air-o-1pst'
[10:54:08][C][wifi:446]:   Signal strength: -58 dB ▂▄▆█
[10:54:08][C][wifi:455]:   Channel: 1
[10:54:08][C][wifi:455]:   Subnet: 255.255.255.0
[10:54:08][C][wifi:455]:   Gateway: 192.168.1.1
[10:54:08][C][wifi:455]:   DNS1: 192.168.1.1
[10:54:08][C][wifi:455]:   DNS2: 0.0.0.0
[10:54:08][C][logger:211]: Logger:
[10:54:08][C][logger:211]:   Max Level: DEBUG
[10:54:08][C][logger:211]:   Initial Level: DEBUG
[10:54:08][C][logger:217]:   Log Baud Rate: 115200
[10:54:08][C][logger:217]:   Hardware UART: USB_SERIAL_JTAG
[10:54:08][C][logger:224]:   Task Log Buffer Size: 768
[10:54:08][C][i2c.idf:083]: I2C Bus:
[10:54:08][C][i2c.idf:084]:   SDA Pin: GPIO7
[10:54:08][C][i2c.idf:084]:   SCL Pin: GPIO6
[10:54:08][C][i2c.idf:084]:   Frequency: 400000 Hz
[10:54:08][C][i2c.idf:094]:   Recovery: bus successfully recovered
[10:54:08][I][i2c.idf:104]: Results from bus scan:
[10:54:08][I][i2c.idf:110]: Found device at address 0x59
[10:54:08][C][uart.idf:159]: UART Bus 0:
[10:54:08][C][uart.idf:160]:   TX Pin: GPIO1
[10:54:08][C][uart.idf:161]:   RX Pin: GPIO0
[10:54:08][C][uart.idf:163]:   RX Buffer Size: 256
[10:54:08][C][uart.idf:165]:   Baud Rate: 9600 baud
[10:54:08][C][uart.idf:165]:   Data Bits: 8
[10:54:08][C][uart.idf:165]:   Parity: NONE
[10:54:08][C][uart.idf:165]:   Stop bits: 1
[10:54:08][C][uart.idf:159]: UART Bus 1:
[10:54:08][C][uart.idf:160]:   TX Pin: GPIO21
[10:54:08][C][uart.idf:161]:   RX Pin: GPIO20
[10:54:08][C][uart.idf:163]:   RX Buffer Size: 256
[10:54:08][C][uart.idf:165]:   Baud Rate: 9600 baud
[10:54:08][C][uart.idf:165]:   Data Bits: 8
[10:54:08][C][uart.idf:165]:   Parity: NONE
[10:54:08][C][uart.idf:165]:   Stop bits: 1
[10:54:08][C][template.sensor:022]: Template Sensor 'PM 2.5'
[10:54:08][C][template.sensor:022]:   State Class: 'measurement'
[10:54:08][C][template.sensor:022]:   Unit of Measurement: 'µg/m³'
[10:54:08][C][template.sensor:022]:   Accuracy Decimals: 0
[10:54:08][C][template.sensor:022]:   Device Class: 'pm25'
[10:54:08][C][template.sensor:022]:   Icon: 'mdi:chemical-weapon'
[10:54:08][C][template.sensor:023]:   Update Interval: 30.0s
[10:54:08][C][template.sensor:022]: Template Sensor 'PM 2.5 AQI'
[10:54:08][C][template.sensor:022]:   State Class: ''
[10:54:08][C][template.sensor:022]:   Unit of Measurement: ''
[10:54:08][C][template.sensor:022]:   Accuracy Decimals: 0
[10:54:08][C][template.sensor:022]:   Device Class: 'aqi'
[10:54:08][C][template.sensor:022]:   Icon: 'mdi:air-filter'
[... 200 lines cut ...]
  package_import_url: github://MallocArray/airgradient_esphome/airgradient-open-air-o-1pst.yaml
  import_full_config: false

packages:
  # switching between arduino and ESP-IDF OTA requires
  # https://github.com/esphome/esphome/pull/5535
  # USB/serial required until then.
# board: github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board-arduino.yaml
  board: github://MallocArray/airgradient_esphome/packages/airgradient_esp32-c3_board.yaml
# captive_portal: github://MallocArray/airgradient_esphome/packages/captive_portal.yaml
  pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003t.yaml
# pm_2.5: github://MallocArray/airgradient_esphome/packages/sensor_pms5003t_extended_life.yaml
  co2: github://MallocArray/airgradient_esphome/packages/sensor_s8.yaml
  tvoc: github://MallocArray/airgradient_esphome/packages/sensor_sgp41.yaml
  airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_esp32-c3.yaml
  hardware_watchdog: github://MallocArray/airgradient_esphome/packages/watchdog.yaml
  wifi: github://MallocArray/airgradient_esphome/packages/sensor_wifi.yaml
  uptime: github://MallocArray/airgradient_esphome/packages/sensor_uptime.yaml
```

</details>

Commenting out `sensor_pms5003t.yaml` and switching back to `sensor_pms5003t_extended_life.yaml` solves the issue for me. Since you fixed https://github.com/MallocArray/airgradient_esphome/issues/158 (super quickly btw, thanks a lot!), I am now fine with the extended life package again, all good, but I thought it's worth mentioning this anyway.

Happy to switch back to the non-extended-life package to collect further logs if needed.

MallocArray, 2025-07-04:
Thank you for reporting it.  Unfortunately I'm unable to reproduce.  With my OpenAir with the non extended package I'm still getting data returned, but I see in your logs that you aren't.  

There were some changes to the pmsx003 sensor in recent ESPHome releases and for some users, a full power cycle is needed to get the device to start responding.
https://github.com/esphome/issues/issues/7040

I only experienced it when testing the PR mentioned in this branch to fix the updated interval not working as expected, and I certainly did have to do the full power cycle
https://github.com/esphome/issues/issues/7063

I haven't needed to do it so far with the main branch, but some people did mention it, so you might try that.  Also, if you didn't see the note from a few months ago, with the change to IDF over Arduino, you need to do one install over USB so it can fully format the internal storage. I don't think it is related to your issue, but just to make sure you are fully setup and eliminate that from being a contributing factor.

Qhilm, 2025-07-04:
ok thanks, I think I can close this then.

btw I was the one who [reported](https://github.com/MallocArray/airgradient_esphome/issues/109#issuecomment-2618596070) the Arduino-to-IDF issue and I keep waiting for the capability to migrate from Arduino to IDF OTA but although the code seems to be there, it is still not [merged](https://github.com/esphome/esphome/pull/5535) into the main branch yet.

closed 2025-07-04 by Qhilm