MallocArray/airgradient_esphome #166: Address ArduinoJson deprecation warnings

MallocArray, 2025-09-04:
In the package for API upload for devices with dual channel PM2.5 sensors, the way the JSON payload body is created is using deprecated options and should be updated to support the new methods before these go unavailable

```
/config/.esphome/packages/76b4402f/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml: In lambda function:
/config/.esphome/packages/76b4402f/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml:23:7: warning: 'template<unsigned int N> class ArduinoJson::V742HB22::StaticJsonDocument' is deprecated: use JsonDocument instead [-Wdeprecated-declarations]
   23 |                 StaticJsonDocument<1024> doc;
      |       ^         ~~~~~~~~
In file included from .piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson.hpp:65,
                 from .piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson.h:9,
                 from src/esphome/components/json/json_util.h:11,
                 from src/esphome/core/string_ref.h:11,
                 from src/esphome/core/application.h:13,
                 from src/esphome/components/api/api_frame_helper.h:12,
                 from src/esphome/components/api/api_connection.h:5,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
.piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson/compatibility.hpp:63:58: note: declared here
   63 | class ARDUINOJSON_DEPRECATED("use JsonDocument instead") StaticJsonDocument
      |                                                          ^~~~~~~~~~~~~~~~~~
/config/.esphome/packages/76b4402f/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml:39:51: warning: 'ArduinoJson::V742HB22::JsonObject ArduinoJson::V742HB22::JsonDocument::createNestedObject(TChar*) [with TChar = const char]' is deprecated: use doc[key].to<JsonObject>() instead [-Wdeprecated-declarations]
   39 |                 JsonObject channels = doc.createNestedObject("channels");
      |                             ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
In file included from .piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson.hpp:42:
.piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:376:14: note: declared here
  376 |   JsonObject createNestedObject(TChar* key) {
      |              ^~~~~~~~~~~~~~~~~~
/config/.esphome/packages/76b4402f/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml:41:58: warning: 'ArduinoJson::V742HB22::JsonObject ArduinoJson::V742HB22::JsonObject::createNestedObject(TChar*) [with TChar = const char]' is deprecated: use obj[key].to<JsonObject>() instead [-Wdeprecated-declarations]
   41 |                 JsonObject channels_1 = channels.createNestedObject("1");
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
In file included from .piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson.hpp:39:
.piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:211:14: note: declared here
  211 |   JsonObject createNestedObject(TChar* key) {
      |              ^~~~~~~~~~~~~~~~~~
/config/.esphome/packages/76b4402f/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml:49:58: warning: 'ArduinoJson::V742HB22::JsonObject ArduinoJson::V742HB22::JsonObject::createNestedObject(TChar*) [with TChar = const char]' is deprecated: use obj[key].to<JsonObject>() instead [-Wdeprecated-declarations]
   49 |                 JsonObject channels_2 = channels.createNestedObject("2");
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
.piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:211:14: note: declared here
  211 |   JsonObject createNestedObject(TChar* key) {
      |              ^~~~~~~~~~~~~~~~~~
```

pyrodex, 2025-10-16:
This got worse with 2025.10.X release, won't even compile now.

```
Compiling .pioenvs/ag-open-air-o-1ppt/lib8a0/noise-c/protocol/cipherstate.o
/data/packages/76b4402f/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml: In lambda function:
/data/packages/76b4402f/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml:22:7: error: 'String' was not declared in this scope
   22 |                 String jsonString;
      |       ^     
/data/packages/76b4402f/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml:23:7: warning: 'template<unsigned int N> class ArduinoJson::V742HB22::StaticJsonDocument' is deprecated: use JsonDocument instead [-Wdeprecated-declarations]
   23 |                 StaticJsonDocument<1024> doc;
      |       ^         ~~~~~~~~
In file included from .piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson.hpp:65,
                 from .piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson.h:9,
                 from src/esphome/components/json/json_util.h:12,
                 from src/esphome/core/string_ref.h:11,
                 from src/esphome/core/application.h:13,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_connection.h:5,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
.piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson/compatibility.hpp:63:58: note: declared here
   63 | class ARDUINOJSON_DEPRECATED("use JsonDocument instead") StaticJsonDocument
      |                                                          ^~~~~~~~~~~~~~~~~~
/data/packages/76b4402f/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml:39:51: warning: 'ArduinoJson::V742HB22::JsonObject ArduinoJson::V742HB22::JsonDocument::createNestedObject(TChar*) [with TChar = const char]' is deprecated: use doc[key].to<JsonObject>() instead [-Wdeprecated-declarations]
   39 |                 JsonObject channels = doc.createNestedObject("channels");
      |                             ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
In file included from .piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson.hpp:42:
.piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:376:14: note: declared here
  376 |   JsonObject createNestedObject(TChar* key) {
      |              ^~~~~~~~~~~~~~~~~~
/data/packages/76b4402f/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml:41:58: warning: 'ArduinoJson::V742HB22::JsonObject ArduinoJson::V742HB22::JsonObject::createNestedObject(TChar*) [with TChar = const char]' is deprecated: use obj[key].to<JsonObject>() instead [-Wdeprecated-declarations]
   41 |                 JsonObject channels_1 = channels.createNestedObject("1");
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
In file included from .piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson.hpp:39:
.piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:211:14: note: declared here
  211 |   JsonObject createNestedObject(TChar* key) {
      |              ^~~~~~~~~~~~~~~~~~
/data/packages/76b4402f/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml:49:58: warning: 'ArduinoJson::V742HB22::JsonObject ArduinoJson::V742HB22::JsonObject::createNestedObject(TChar*) [with TChar = const char]' is deprecated: use obj[key].to<JsonObject>() instead [-Wdeprecated-declarations]
   49 |                 JsonObject channels_2 = channels.createNestedObject("2");
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
.piolibdeps/ag-open-air-o-1ppt/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:211:14: note: declared here
  211 |   JsonObject createNestedObject(TChar* key) {
      |              ^~~~~~~~~~~~~~~~~~
Compiling .pioenvs/ag-open-air-o-1ppt/lib8a0/noise-c/protocol/dhstate.o
/data/packages/76b4402f/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml:58:26: error: 'jsonString' was not declared in this scope
   58 |                 serializeJson(doc, jsonString);
      |                          ^~~~~~~~~~
```

MallocArray, 2025-10-17:
I see the same.  It has to do with the way the JSON data is structured to send up to the AirGradient servers, but I didn't write the block for this particular one with dual PM2.5 sensors.  I got it from another user, so I don't know for sure how to update it to whatever the newest format is, but I'll see if I can get Copilot to give a hand.

pyrodex, 2025-10-17:
> This compiles, but I don't have one of this particular model to test fully.
> 
> Can you change your package line to this to test the fix?
> 
> ```
>   airgradient_api: github://MallocArray/airgradient_esphome/packages/airgradient_api_esp32-c3_dual_pms5003t.yaml@ppt_arduinojson_fix
> ```

Just compiled it with no issues and warnings. Checked their console and it shows it reporting home.

MallocArray, 2025-10-18:
Addressed in https://github.com/MallocArray/airgradient_esphome/releases/tag/5.3.3